sinara-hw/zapper #1: [RFC] new EEM: zapper (high-voltage source/PZT driver)

dtcallcock, 2019-02-06:
High voltage supplies are commonly required in ion trap experiments.  These are widely commercially available but typical models have many of the following drawbacks:

- Noisy output
- Radiated RFI from the box
- Expensive
- Poor stability
- Low channel density
- High power dissipation when voltage output static
- Unnecessarily high current capability (safety issue as well as being the root cause of most of the above)
- Flaky USB connection with high latency

# Use cases

### Trap electrode voltages

Zotino only provides ±10V.  This is generally fine for small high-electrode-density surface traps.  However many groups operate larger surface traps or macroscoptic traps of the 'blade' or 'wafer' variety.  These typically need much higher voltages, from a few 10s of Volts up to a few hundred Volts.  Requirements are basically the same as for Zotino but scaled to higher voltages.  The [thesis](https://www2.physics.ox.ac.uk/sites/default/files/2011-08-15/cjbthesisfinal_pdf_33239.pdf) of Chris Ballance has a good discussion of the requirements and a circuit design that could be the basis of this EEM.  This type of trap generally has less electrodes so a <32-ch solution may be acceptable.  

### Slow piezos

We have several piezo-actuated mirrors on our experiment, such as the Thorlabs [Polaris K1S2P](https://www.thorlabs.com/newgrouppage9.cfm?objectgroup_ID=5035).  These need 0-150V and have a 0.35uF of capacitance.  The Thorlabs [MDT694B/MDT693B](https://www.thorlabs.com/newgrouppage9.cfm?objectgroup_id=1191) are ubiquitous in our lab but suffer many of the problems described above.  For this use case, the new [PiezoDrivePD32](https://www.piezodrive.com/product/pd32-multi-channel-piezo-driver/) may be a good commercial alternative when coupled with a Zotino.  I haven't tested one out but we have been happy with other products from PiezoDrive. 

# Implementation

### DAC

16-bit resolution.  

A few options:

- Use Zotino.  The card would sit next to Zotino and be connected to the IDC headers on it via a ribbon cable(s).  Lower risk, less development work, and the Artiq driver would be pretty much for free.  Frees up board real estate as no EEM connector etc. would be needed.

- Use an onboard DAC.  The 32-ch AD5372 in Zotino has 8-ch ([AD5362](http://www.analog.com/en/products/ad5362.html)) and 16-ch ([AD5360](http://www.analog.com/en/products/ad5360.html)) equivalents if that is likely to reduce risk/effort.  At 4-ch and 8-ch, more accurite, lower noise DACs are available.

### Channel Count

This will probably depend a lot on the power consumption, size, and cost of the output stage we decide on.  8 would be a reasonable starting point.

### Ranges

Traps generally require a mix of high positive voltages for endcaps (becasue our ions are positively charged) and lower bipolar voltages for other electrodes.  There seems to be 2 options for dealing with this:

- Per board range selection.  The board could be configured as either single ended or bipolar.  If different ranges are required, you just use two boards.  The cheaper the board is and the lower the number of the channels, the less of an issue this is.

- Per channel range selection.  Board has three supply rails (+V1, +V2 and -V2).  Each channel can be changed to run at 0 to +V1 or -V2 to V2 using jumpers or component selection.

### Voltages

It would be good to hear from potential users here.  As a starting point, how about:

0 to +V1 or -V2 to +V2 where V1= 200, 100, or 50V and V2 = 100, 50 or 25V

Historically some traps have used >1kV but my guess is that there isn't a sensible design that covers that and 10s of volts.

### Bandwidth

Zotino has 75kHz analog bandwidth.  This sounds like a reasonable starting point.

There should be the option to install components that drastically limit this to say, 100s of Hz.

### Stability

Zotino output stage has "Worst-case OpAmp self-heating is about 25C. At 0.2ppm/C, this is 4ppm, which is fine!"  This might be harder to achieve at high voltages but <10ppm would be nice.

### Connectors

- Individual Front Panel coax connectors - Often slightly less common connectors are used on HV supplies to prevent accidentaly connecting the wrong thing and destroying it.  I have seen SMB, SMC, LEMO OO and LEMO 0B used.

- High Density Connector - D-sub etc. 

- Both - Same idea as Zotino, where the high density connector can be installed or a ribbon cable run to a panel with individual connectors on it (ie. a [BNC-IDC](https://github.com/sinara-hw/BNC_IDC/)).  We could make a version of BNC-IDC with SMB etc. instead.

hartytp, 2019-02-06:
Neat. What's the plan? Will you use a potted module? HV OpAmps? Or discrete components? Will it be something like the JQI design?

gkasprow, 2019-02-06:
Let's try to build a discrete amplifier circuit. HV power supply could be COTS component.

hartytp, 2019-02-06:
Cool, it's a fair bit of work that way, but if you get it right the performance is really good. Also, more fun :)

IIRC, the JQI design is based from a laser printer IC which provides the HV supply. Then they add a linear circuit to do post regulation. I think that's probably the best way to go.

gkasprow, 2019-02-06:
do you mean [this](http://sci-hub.tw/10.1063/1.4969059) design?

hartytp, 2019-02-06:
That's the one. I've never used it, but it seems nice. Probably worth asking @npisenti for his views on what they like/dislike about the circuit and any ideas they have for improvements.

A few channels of something like that on an EEM would be ace. Maybe PoE?

There are a few design decision you need to make like modulation sources (analog? digital? EEM?) bandwidth etc.

hartytp, 2019-02-06:
For the post regulator, might also being worth checking out the design that @cjbe posted a while ago.

IIRC there are (as always) some nice designs for this in the Hobbs book (and I remember seeing some designs for comically low noise PZT drivers in various places he lurks on the internet).

JakeBlackmore, 2019-03-18:
Hello,
I was directed here by a colleague using ARTIQ. Is there scope for a version of this design which works up to +/- 5 kV?

gkasprow, 2019-03-18:
The specification is not yet fully defined. What BW and slew rate do you need? What load capacitance? What did you use before?

gkasprow, 2019-05-31:
I have an application where a lot of channels is needed, 8 or even 16 per module, +/- 150V at least with reference to GND and ultra-low noise, a few uV pp. Bandwidth is not that critical, a few kHz does the job. Load capacitance is not critical, I will have tens of pF. 
Such a driver could be based on Zotino to keep the SW compatibility to some extent.
Can we compile the final specification for my student which also includes specification above?

hartytp, 2019-05-31:
> Such a driver could be based on Zotino to keep the SW compatibility to some extent.

I'm fine with keeping the HV amp and DACs on separate boards if you prefer. However, FWIW, my feeling is that having fewer boards in the stackup is always better. It's trivial to add a multi-channel DAC + filters to your HV amb board, so not much time saved by using Zotino for this. (However, if you use a DAC in the same family as the Zotino DAC then you can reuse the same design and software, so it's just a copy-paste job).

gkasprow, 2019-05-31:
Oh, I meant modifying the Zotino design :)
I want to have a single multi-channel board that has HV, DAC, and drivers on the same PCB.
Would the Zotino DAC speed be enough for your application? Or it would be better to rather switch to Fastino type DACs?

hartytp, 2019-05-31:
> Oh, I meant modifying the Zotino design :)

Aah, glad we agree :)

> Or it would be better to rather switch to Fastino type DACs?

That would be nice as it allows things like locking cavities at a decent bandwidth. The downside is that doing that with a decent number of channels makes the digital interface more complex (misoc/artiq doesn't support QSPI right now).

gkasprow, 2019-05-31:
well, we can make the SPI interface first, then upgrade the FPGA. The same we will probably do for Fastino.

gkasprow, 2019-06-01:
What about the following spec:
- channel count: 8
-  low [cost Apex amplifiers](https://www.digikey.com/products/en/integrated-circuits-ics/linear-amplifiers-instrumentation-op-amps-buffer-amps/687?FV=a4c01f5%2Ca4c0208%2Ca4c0277%2Ca4c0279%2Ca4c027a%2Ca4c027c%2Ca4c027d%2Ca4c0293%2Ca4c0294%2Ca4c02a8%2Cffe002af%2C1f140000&quantity=0&ColumnSort=1000011&page=1&pageSize=25), PA341DF would be rpobably fine
- 8 channel AD5362
- at each output we can install active AC-coupled noice cancellation circuit that can be disabled by software when high speed operation is required. We can use OPTOMOS relays to diasable it.
- operating voltage could be controlled by DIP switches. Negative -50, -100, -200, positive +50, +100, +200 with limitation of 350V with HW protection against opamp damage. DAC references would need to be adjusted as well.
- COTS DC/DC converters
- what about the offset cancellation ADC?
- EEM interface. It can be always controlled by Stabilizer EEM and supplied by its PoE.
- DSUB connector + on-board IDC connector for BNC breakout.

gkasprow, 2019-06-01:
By noise cancellation circuit I meant [this](https://github.com/JQIamo/hv-piezo-driver/blob/master/Eurocard-HV-piezo-driver_Schematic_v1.2.pdf) approach

gkasprow, 2019-06-01:
Two such [DC/DC](https://recom-power.com/pdf/Innoline/Rxx-B.pdf) converters could do the job. They go down to 90V but could not find others that can be adjusted more than 90...200V

restelli, 2019-07-11:
> For everyone considering PDu150, the outputs are differential only, NOT single ended, and you cannot ground-reference one output. Thus it won't work for typical piezo actuators where the signal is ground-referenced on one lead.

Neil Pisenti and I a few years ago designed a ground-referenced low noise piezo driver for the JQI strontium lab based of the [DRV2700](http://www.ti.com/product/DRV2700). Despite the DRV2700 being quite noisy by itself we were able to suppress its ripple and obtain noise levels comparable with the best commercial controllers. Noise characterization and overall design are available in our RSI paper: https://arxiv.org/abs/1609.03607

The project is on Github: https://github.com/JQIamo/hv-piezo-driver

The main limitations of the design is that the control bandwidth is limited for large amplitude signals.

sbourdeauducq, 2019-08-07:
If we have a lot of bandwidth (1MHz or more), then this circuit can also be useful with EOMs (e.g. https://www.thorlabs.com/newgrouppage9.cfm?objectgroup_id=2729) for laser intensity stabilization.

dtcallcock, 2019-08-08:
Note that the standard drivers for these are +/- 200V, 100mA beasts.  If that's what you are after it may be incompatible with (my hope of) something that's higher channel density, lower noise, and lower power consumption than what's already out there.

sbourdeauducq, 2019-08-08:
200V*1MHz*14pF = 2.8mA. Are the standard drivers able to go much higher than 1MHz, is the 14pF specification inaccurate, or is there something else going on?

dtcallcock, 2019-08-08:
Don't forget that coax cable is some 10s of pF/ft.  

Still, their driver does seem a bit over-specced.  Also somewhat overpriced at $2.6k...

jordens, 2019-08-08:
And there is a factor of 2*pi > 6 as well.

adamkolodynski, 2020-01-23:
Hello
I have a few questions about the design of this board.
1. Are onboard DAC required? Using Zotino would make it way simpler, and there is always a possibility to add e.g. fast DAC later, in another version.
2. What power supply is available, except the EEM connector?

dtcallcock, 2020-01-23:
Hi #adamkolodynski

Are you developing the spec Greg laid out [above](https://github.com/sinara-hw/zapper/issues/1#issuecomment-497932589)?

If so:

> Are onboard DAC required? Using Zotino would make it way simpler

Sure, but Zotino is quite expensive.  Having to add a pricy 32ch card to use this card would be a bit annoying.  Using AD5362 (8ch version of AD5372 32ch DAC on Zotino) should allow a lot of Zotino design/code reuse.

> there is always a possibility to add e.g. fast DAC later, in another version.

If the analog BW is <100kHz, then the AD5362 should be able to pretty much saturate that.  There could also be an IDC input connector on the board and jumpers to disconnect the DAC to allow this.

> What power supply is available, except the EEM connector?

If it has DACs on it and power consumption is reasonable then I think just EEM is fine and in line with what's done elsewhere.

gkasprow, 2020-01-23:
About the power supply selection, look [here](https://github.com/sinara-hw/Stabilizer_Piezo_Driver/issues/4)

JakubKajzer, 2020-02-01:
Does anybody need a noise cancellation circuit in this module?

adamkolodynski, 2020-02-21:
After reading the thread about power supply I see some major issues. 
Can we use ADHV4702 with -12V and +110V supply, or keep PA341DF? 
Is +110V enough range in this use? 
ADHV4702 has 20 mA output current, what load we have to drive if we still target 100 kHz?

sbourdeauducq, 2020-02-21:
PA341 is "No longer supported for design in" and no longer supplied by DigiKey.
FWIW I've also had a bad experience with an Apex part that was expensive and not performing so well; I would trust ADI more or even a design with discrete BJTs (not so hard to do IME and you can exceed the performance of commercial ICs - but BJTs with the right characteristics can be subject to sourcing problems as well).

gkasprow, 2020-02-22:
I'd give an option to adjust the positive HV value.
The negative would be -5V, the positive could be 105 or 205V. The IC requires 2V supply margin.
The power supplies need to have LC filter and LDOs at its outputs.

gkasprow, 2020-02-22:
Before you publish the entire schematic, please open separate issues with proposed parts of schematics like decoupling, DAC, supply, etc It will make the feedback faster.

adamkolodynski, 2020-03-31:
I started 2 new issues. First about DAC, and second about the power supply, and will probably add another about the output stage.

gkasprow, 2020-07-16:
can you please commit the current version of schematics?

adamkolodynski, 2020-07-20:
I have a few questions about connectors.
One I assume will be IDC to connect with this [IDC-BNC adapter](https://github.com/sinara-hw/BNC_IDC/wiki). I don't know the purpose of SCSI power pins(+-13V) here, should I left them floating?
The second will be standard D-SUB connector, and since we don't use differential output just reference to GND and low currents can we use 15 pins D-SUB with 8 signal pins and 7 GND or simply 25 pins D-SUB with 8/8 pins configuration and rest unconnected.
[Zapper.pdf](https://github.com/sinara-hw/zapper/files/4950283/Zapper.pdf)

gkasprow, 2020-07-20:
I wouldn't use the same connectors as Zotino. Sooner or later someone will mix the two and cause damage.
Just use the D-SUB connector. High-density D-SUBs may not like the 200V range. Just use standard 15pin D-sub. Maybe there are some standard cables available that could be used as they are? If so, it would be nice to follow their pinout...

adamkolodynski, 2020-08-07:
I found standard 15 pin D-sub cable and it belongs to GAMEPORT cable(obsolete standard for joystick). Is it worth following this pinout(picture below): signals to analog and digital pins, gnd to gnd and +5V?
[image]
About breakout to BNC from DA-15, is it required to create an adapter similar to existing [link](https://github.com/sinara-hw/BNC_IDC/wiki) ?

gkasprow, 2020-08-12:
VGA  have some lines shielded, some not, so it is not good idea.
The Gameports cables were attached to joysticks, I'm not sure if one can buy them.
I was thinking about DB15 serial cables like this [one ](https://www.amazon.com/Your-Cable-Store-10-DB15/dp/B002HYJS7I)

adamkolodynski, 2020-08-14:
I meant exactly cable like this, just I found it described as gameport to gameport, anyway they are hard to find, so I will leave pinout like it is now. 
I made some minor changes(resistor tolerance, track width, etc.) to the first release. I already asked @gkasprow as my supervisor for a more complex check, but I would appreciate any feedback.