sinara-hw/zapper #3: Zapper - 8 channel DAC and EEM

adamkolodynski, 2020-02-28:
### DAC voltage range
I think the best output range would be 0 to+10 V, [DAC](https://www.analog.com/media/en/technical-documentation/data-sheets/AD5362_5363.pdf) has also configurable offset registers, but I don't know how convenient are their usage. The voltage range is defined by 4xVref, but we have to be in +-1.4 V from the power supply. I see two options:
-	Power supply -5 V/+12 V, Vref=2,5 V, but we need to use offset registers.
-	Power supply +-12 V, Vref =5 V, but we losing half of the span.

### DAC additional functions
Translating signal from EEM and DAC design are very similar to Zotino, but [AD5362](https://www.analog.com/media/en/technical-documentation/data-sheets/AD5362_5363.pdf) has more additional functions that AD5372, so I need to know which can be useful. 
1. Packet Error Check Output - now connected to EEM connector, but if any more important use of this I/O appears probably will leave it.
2. Monitor - output could be connected to a gold pin(or another connector) and used for testing and monitoring.
3. Temperature sensor - might be useful later.
4. Programmable GPIO - will find some application later.

### Current schematic files

[LVDS schematic](https://github.com/sinara-hw/zapper/files/4266615/LVDS.pdf)
[DAC_8CH.pdf](https://github.com/sinara-hw/zapper/files/4266616/DAC_8CH.pdf)

gkasprow, 2020-04-21:
@jordens do you think we could need the PEC output?
Yes, add a standard test pad.
we don't expect the DAC to overheat.

please connect the VREF pins.

jordens, 2020-04-21:
not critical

adamkolodynski, 2020-04-27:
Added test pads and VREF: [DAC_8CHv2.pdf](https://github.com/sinara-hw/zapper/files/4539184/DAC_8CHv2.pdf)
How about the offset voltage calibration similar to this [Automated Calibration](https://www.analog.com/media/en/technical-documentation/application-notes/AN-1582.pdf)

gkasprow, 2020-05-01:
I think the offset will be dominated by the output stage offset anyway. No real reason to calibrate the DAC only.

adamkolodynski, 2020-07-20:
I found that AD5362BCPZ run of out stock in most distributors(except Chinese sites which is hard to verify), so I changed it to AD5362BSTZ which is also hard to buy but still available.

gkasprow, 2020-07-20:
Use two footprints and assembly options as we do in Zotino

adamkolodynski, 2020-07-20:
In Zotino DAC packages have the same pin number and pin description and here one has 52 pins and the second has 56 pins. Their description and placements are not fitting in many cases. I don't think it is possible here.

gkasprow, 2020-07-20:
U can try to place them side-by-side. We don't have high-speed signals here.