sinara-hw/zapper #4: Zapper - power supply

adamkolodynski, 2020-03-30:
### [Zapper](https://github.com/sinara-hw/zapper/issues/1) power supply
In previous threads about zapper, it was mentioned that the most fitting output voltage range would be 0-200 V. ADHV4702-1 accepts 220 V supply(maximum rating mention 225 V), so we need at least -5 V/205 V(unless 190/195 would work too).

The lowest possible quiescent current per channel is 0,6 mA(with 100k Ohm resistor in ADHV 4702-1) but reduces dynamic performance, so I think 50k Ohm will be better solution and reduction to 1 mA is enough.
I found some DC/DC converter which fit required >200V or eventually 200V, but I need to know how much current per channel we need on output, and if we need a switchable operating range. Below there are some converters which I find useful.

1. [SF_F_Series](https://github.com/sinara-hw/zapper/files/4403075/SF_F_Series.pdf) or [SF_FS_Series](https://github.com/sinara-hw/zapper/files/4403076/SF_FS_Series.pdf) 10 W converters with regulation 0 to 200 V, or 0 to 300 V, but quite expensive(150$ F series, 180$ FS). The output is proportional to the input but heavily depends on the load.
2. [VG10 Series ](https://github.com/sinara-hw/zapper/files/4403168/VG10.Series.Full.Specification.pdf) -10 W, 250 V output voltage, but only 90$
3. [Rxx-B](https://github.com/sinara-hw/zapper/files/4403202/Rxx-B-334986.pdf) - 5 W, regulated from 92 V to 200 V, around 40$
4. American Power Desing has few models which could fit, but they are no available in Mouser/Digikey only from one distributor in Europe.

JakubKajzer, 2020-04-02:
If the output current of ADHV4702-1 is too low, we can use [PA441](https://www.apexanalog.com/resources/products/pa441u-pa443u.pdf) (next generation of PA341), it has 60mA continuous output current and 120mA in peak and 350V max voltage supply.

For a negative voltage supply (if we keep -5V), I think we can use COTS modules like this:

1.  [NMJ Series](https://www.farnell.com/datasheets/2625591.pdf) from Murata - 100mA, 60% efficiency
2. [TES 2N Series](https://www.tracopower.com/products/tes2n.pdf) from Traco Power - 200mA, 73% efficiency
3. [R1D series](https://pl.mouser.com/datasheet/2/468/R1S_R1D-1709880.pdf) from Recom - 100mA, >72% efficiency

They are symmetrical voltage converters, so maybe somewhere positive 5V will be useful.
For low output current, the ripple and noise can be significant, so it must be well filtrated.

adamkolodynski, 2020-04-04:
Output current is mostly limited by EEM connector, and according to the [wiki](https://github.com/sinara-hw/meta/wiki/EEM) board can draw 2A@12V, so 120mA@200V is impossible even with a single channel.

About negative voltage supply, I think about symmetrical voltage converter 5V, where -5 V will be connected to LDO(-4,5V), and +5 to 3,3V regulator to supply LVDS translation(around 400mA). However this requires converter with a higher power, and LDO 5->3,3 mean losing significant power.

JakubKajzer, 2020-04-04:
ADHV4702-1 have 20mA max output current. If we keep 8 channels as is mentioned in the thread about zapper it can draw 160mA@200V at one moment. So probably we need another power supply input or lower channel density.

About the 3V3 supply, I think that instead of LDO, we can take TPS62175DQC like in Zotino, but the input range of this IC is 4.75V to 28 V so it's quite near 5V and it can be less efficient than LDO.

gkasprow, 2020-04-21:
The output is purely capacitive. We will consume such current only during voltage change. So don't worry about power suply current

gkasprow, 2020-04-21:
Copy the 3V3 converter design from other designs. For -5V you can use small 1W power converter

adamkolodynski, 2020-05-14:
I added 3V3, 11,5 V and -4,5 V. How about using a cascade of [R12-100B](https://recom-power.com/pdf/Innoline/Rxx-B.pdf) to provide HV? [PowerSupp.pdf](https://github.com/sinara-hw/zapper/files/4628496/PowerSupp.pdf)
With two converters in series we have enough output power(5W is not enough) but doubled ripple(max. 200 mVpp). With higher voltage, we can add a voltage regulator and an LC filter.
Only one with such wide input voltage is [LR8](https://www.mouser.com/datasheet/2/268/20005399A-909415.pdf) but it has too low maximum current.
Maybe we can use floating LDO for example [LT3085](https://www.analog.com/media/en/technical-documentation/data-sheets/3085fb.pdf) or [LT3080](https://www.analog.com/media/en/technical-documentation/data-sheets/3080fc.pdf) or even LM317.

JakubKajzer, 2020-05-20:
How about using one R12-150B which can go up to 200V and have 5W output power? 1.6W goes to supply quiescent current. We can add a bulk capacitor (max is 40uF for R12-150B) to have additional charge and we still have 3.4W to charge or discharge piezoelectric.

dtcallcock, 2020-05-20:
Will you add an option to switch the HV rail to say +110V?  100V is sufficient for many piezo applications and this would allow power dissipation to be reduced and eliminate the risk of accidentally overdriving the piezo.

adamkolodynski, 2020-05-20:
If you say it's necessary, R12-150B can be regulated +200 V to +92 V with a single switch, so two ranges 200V and 110V are possible.
Using an electrolytic bulk capacitor is the option if we want to use a single 5 W converter, but we don't have an extra voltage range for LDO so the ripple will be more significant.

dtcallcock, 2020-05-20:
> If you say it's necessary, R12-150B can be regulated +200 V to +92 V with a single switch, so two ranges 200V and 110V are possible.

This is also true with cascaded R12-100B as they have a 50-135V range.

> if we want to use a single 5 W converter, but we don't have an extra voltage range for LDO so the ripple will be more significant.

How much of this ripple is likely to end up on the output given the PSRR of the opamp?  If a reg is needed then I think a 180V max output for example isn't the end of the world.

adamkolodynski, 2020-05-20:
> This is also true with cascaded R12-100B as they have a 50-135V range.

In datasheet Vout range of R-12-100B is limited/recommended to 190V-260V.
[image]

I asked RECOM technical support about this, and they answered:

[image]

> How much of this ripple is likely to end up on the output given the PSRR of the opamp? If a reg is needed then I think a 180V max output for example isn't the end of the world.

Max. ripple is 100mVpp on 20MHz bandwidth, with 200kHz switching frequency. ADHV4702-1 have around 30 dB +PSSR for 200kHz. Usually, regulators don't have high ripple rejection in higher frequencies so it's not necessary to add one.

dtcallcock, 2020-05-21:
> In datasheet Vout range of R-12-100B is limited/recommended to 190V-260V.

Sorry, I missed that section.

> Usually, regulators don't have high ripple rejection in higher frequencies so it's not necessary to add one

So perhaps just use the R12-150B with passive filtering to augment the opamp PSRR to reduce the ripple below the noise floor?

adamkolodynski, 2020-06-05:
I added R12-150B with a panel switch to change from +200 V to +110 V. LC filter should give around -38 dB on 200 kHz to reduce output ripple. I used a film capacitor with low ESR so I added 2 Ohms resistor to avoid peak at the resonance frequency.

[PowerSupp2.pdf](https://github.com/sinara-hw/zapper/files/4735527/PowerSupp2.pdf)

dtcallcock, 2020-06-05:
- I think there is a missing resistor between VREF and VADJ.

- Should there be a pi filter on the input of the switcher to help prevent conducted EMI going back into the rest of the crate?  For example, here's the input filter from Fastino:

[image]

> a panel switch to change from +200 V to +110 V

Is hot switching undesirable?  If so, it may be better to put it down on the PCB to discourage that (or accidental switching in general).  I don't think it's something people will need regular access to.

adamkolodynski, 2020-06-10:
>I think there is a missing resistor between VREF and VADJ.

VREF has a 1k resistor already unless it means something else(datasheet below). Maybe I should show it somehow on the schematic.
[image]

> Should there be a pi filter on the input of the switcher to help prevent conducted EMI going back into the rest of the crate? For example, here's the input filter from Fastino:

Yes, should I add a separate filter to the HV converter, and second for rest converters or single pi filter is enough?

>Is hot switching undesirable? If so, it may be better to put it down on the PCB to discourage that (or accidental switching in general). I don't think it's something people will need regular access to.

I will change the switch to PCB mounted. Even if hot switching shouldn't be an issue, changing range shouldn't be used very frequently.