sinara-hw/zapper #5: Zapper - Opamp

JakubKajzer, 2020-04-16:
# [Zapper](https://github.com/sinara-hw/zapper/issues/1) - Opamp

## Input
### Low pass filter
At the input, I've decided to make a two-stage low-pass filter to decrease noise bandwidth at ADHV4702-1 input. Filter has 100kHz 3dB bandwidth, I've chosen that frequency because it's close to max frequency of DAC. 
### Clamping diodes
Clamping diodes protect the device from overheating when operating continuously at or near full power bandwidth but reduce dynamic performance (significantly decrease slew rate). I'm not sure if we need them, so I left a placeholder for them.
## Output
### Gain
Gain is set to 20V/V to cover 0-200V from 0-10V DAC.
### Stability
For stability when driving a capacitive load, there is a serial output resistor, parallel resistor to load and a small capacitor between COMP and OUT pin to reduce output stage peaking associated with capacitive loads.
##### Series resistor
47ohm resistor is a value from simulations, for this value, the opamp is stable from 10pF to 100uF capacitance load. But it decrease bandwidth. For 22ohm resistor, for capacitance loads from 10nF to 1uF, gain can increase by 4dB according to simulations. Maybe with COMP capacitor, it will work fine for 22ohm. 
##### Parallel resistor
According to the ADHV4702-1 datasheet, the amplifier is more stable if it has a parallel resistive load to load with capacitance. The datasheet doesn't specify the value of this resistor. In a typical application, they suggested 10k at ±105V output, so for 0-200V at output I think 33k will be okay.
##### COMP capacitor
ADHV4702-1 have external compensation pin. Placing about 5-6pF capacitor can significantly increases the stability while driving capacitive loads. So I choose 5.6pF capacitor.
## Features
### Short-circuit protection
ADHV4702-1 has a shutdown pin (SD) and a temperature sensor pin (TMP). Connecting them and placing 200k resistor to ground will provide short-circuit protection through the thermal shutdown feature.
### Quiescent current
Quiescent current can be set within a range from 0.6mA to 3.3mA. Low current reduces dynamic performance and increase noise. Reduction to 1mA with a 50k resistor is a balance between low current and low noise.

[Opamp.pdf](https://github.com/sinara-hw/zapper/files/4486564/Opamp.pdf)

gkasprow, 2020-04-21:
Please add some signalization of the overtemp condition. I'd add OR function of all TEMP signals and connect to the front panel LED

JakubKajzer, 2020-05-10:
I added a Schmitt trigger to show the actual state of the opamp (on/off). 
CD4078BMT (8-input) OR gate will be perfect for this application.

[Opamp.pdf](https://github.com/sinara-hw/zapper/files/4605438/Opamp.pdf)

dtcallcock, 2020-05-20:
>0.6mA to 3.3mA. Low current reduces dynamic performance and increase noise. Reduction to 1mA with a 50k resistor is a balance between low current and low noise.

A DIP switch to select might be nice here.  Being able to half the board's quiescent dissipation from 1.6W to 0.8W (+ decrease in converter losses) is a nice feature if you don't need the dynamic performance.

> a parallel resistive load to load with capacitance. ... so for 0-200V at output I think 33k will be okay.

This resistor dissipates 1.2W at 200V output.  That seems excessive.

JakubKajzer, 2020-06-18:
> A DIP switch to select might be nice 

The DIP switch has been added. I was thinking about adding two SPST DIP switch to choose 3.3mA, 1mA, or 0.6mA quiescent current but finally, I left it as is.

> This resistor dissipates 1.2W at 200V output. That seems excessive.

I overlooked this. According to the opamp datasheet, it slightly increases stability for unity gain configuration, so it doesn't change much there. I left this resistor as a placeholder.

[Opamp.pdf](https://github.com/sinara-hw/zapper/files/4799377/Opamp.pdf)

adamkolodynski, 2020-07-02:
Hello
I checked the existing schematic and made some changes :
1. You have used double op-amp AD8676 one to lowpass filter and second to the Schmitt trigger. I'm pretty sure u can't use both op-amps with different voltage supply, and you connected two different voltages to V+ pin(nr 8) and V- pin. I changed filter opamp to single-channel AD8675. It has additional null pins but I think when most offset come anyway from ADHV4702-1 there is no point in using them.
[OutputChannel.pdf](https://github.com/sinara-hw/zapper/files/4866971/OutputChannel.pdf)

2. I moved Schmitt trigger to another schematic where I used 4-channel AD8544 which has less quiescent current and saves place(AD8676/75 has not been specified below 10V of supply anyway). I don't think we need 3$ opamp to Schmitt trigger so I used a cheaper one. I changed one resistor in the voltage divider to have more accurate thresholds(1,6V and 0,8V). In the end, I added a LED board indicator that was used before in a few designs (Zotino, Fastino maybe some more).
[SCHMITT_LED.pdf](https://github.com/sinara-hw/zapper/files/4866965/SCHMITT_LED.pdf)

3. In the main thread, it was mentioned zapper could have an AC-coupled noise cancellation circuit. I think we can use zero offset lowpass filter, a similar approach to [LINK](https://www.edn.com/zero-offset-active-lowpass-filter-part-1/) and [LINK2](https://ir.library.louisville.edu/cgi/viewcontent.cgi?article=2077&context=etd) It doesn't affect the DC but reduces bandwidth a lot. Resistor R13 should double the R11 value to achieve the proper Q factor. And the values of the capacitors we can change depends on what resonant frequency we want.
The only thing is how to properly switch it off. One way is to use MOSFET photocoupler which at least 200V Voff, the problem is that they have usually around 20 Ohms ON-state resistance, which is not acceptable, but I found Toshiba relays with lower resistance(max. 2 Ohms). [TLP3825](https://toshiba.semicon-storage.com/eu/semiconductor/product/optoelectronics/photorelay-mosfet-output/detail.TLP3825.html) [TLP3549](https://toshiba.semicon-storage.com/ap-en/semiconductor/product/optoelectronics/photorelay-mosfet-output/detail.TLP3549.html) [TLP3558A](https://toshiba.semicon-storage.com/ap-en/semiconductor/product/optoelectronics/photorelay-mosfet-output/detail.TLP3558A.html) but it will still degrade a bit filter characteristics.
The second option is simply SPST toggle switch but I don't know how much insulation resistance and dielectric strength should it have to work fine with 200VDC in a long time(cold switching).

JakubKajzer, 2020-07-13:
1. 
> I'm pretty sure u can't use both op-amps with different voltage supply

Sorry, my mistake, thanks for finding this.
> I changed filter opamp to single-channel AD8675.

If you decided to use another opamp for Schmitt trigger, why we can keep double opamp, it will occupy less area on PCB, or use even quadruple opamp.

3. 
There is a [issue](https://github.com/sinara-hw/zapper/issues/6) about noise cancellation. Zero offset lowpass filter sounds good, but if we keep R13 as a doubled value of R11, it's hard to achieve tens of hertz passband (if we want to reduce 1/f noise) without having tens of microfarads. If we raise R13 value, then will be a problem with the Q factor.