sinara-hw/zapper #6: Zapper - filter stage

JakubKajzer, 2020-05-20:
# Filter stage
I have designed a switchable low pass filter (3db BW is 5Hz) to reduce noise and bandwidth. I have used two in series capacitors and resistors so they can withstand 200V.  I have used a double photoMOS relay and DIP switch to turn on/off filtration.

[Filter.pdf](https://github.com/sinara-hw/zapper/files/4653900/Filter.pdf)

dtcallcock, 2020-05-20:
Don't you need [balancing resistors](https://www.illinoiscapacitor.com/pdf/Papers/voltage_balancing_resistors.pdf) to add capacitor voltage ratings like that?

You can get [450V rated 1210 C0G caps](https://www.digikey.com/product-detail/en/tdk-corporation/C3225C0G2W223J230AA/445-15463-1-ND/4278636) and [400V rated 0805 resistors](http://www.koaspeer.com/products/resistors/high-voltage/hv73/).  A bit non-standard but still <$1 components.

JakubKajzer, 2020-05-20:
I've checked some Vishay resistors and they were 150V rated, so I decided to place two in series, but you're right, I found [250V rated 0603 capacitors](https://www.mouser.com/ProductDetail/TDK/CGA3E3C0G2E102J080AE?qs=%2Fha2pyFaduicxNOjy8ivRFAas%252B8l6U0GIWyPRbHwgFCnDoTKN0yopa15bWrAGFeU) and [350V rated 0603 resistors](https://www.mouser.com/ProductDetail/ROHM-Semiconductor/KTR03EZPF3303?qs=sGAEpiMZZMvdGkrng054tywGjayRHxH2YGnpUi15%252BKVccrQ3hD12qA%3D%3D) so I'm going to change the schematics.

dtcallcock, 2020-05-20:
The AQW210EHA has a typical Ron of 18Ohms.  This should be considered when selecting the series output resistor for the opamp.

How about using DIP switches directly to avoid this and remove a point of failure?  You could also then use DPDT to disconnect the filter from the output completely.  At these frequencies, parasitics shouldn't cause problems.  They [can't hot switch 200V](http://www.grayhill.com/assets/1/7/DIP_Series_76_and_78_DPDT.pdf) but I think it's fine to require the board to be off when switching.

JakubKajzer, 2020-06-18:
> How about using DIP switches directly to avoid this and remove a point of failure?

I think we don't need to turn on/off filter often, so no possibilities of hot switching shouldn't be a problem.
I've changed also the filter components.

[Filter.pdf](https://github.com/sinara-hw/zapper/files/4799871/Filter.pdf)

gkasprow, 2020-07-16:
@JakubKajzer please commint all the Altium source files.

adamkolodynski, 2020-07-20:
This way of switching off resolve a major problem with zero offset filter I proposed in OPAMP issue. Bypassing the whole filter allows us to place large resistance in series with op-amp output. I used dual ADA4807-1 which requires only 1mA/channel and two 0805 68nF capacitors. In a result filter with the same -3dB frequency has a better slope in lower frequencies(to 1 kHz) and it doesn't kill op-amp output current(time of setting the voltage on capacitive load).
[FilterZeroOffset.pdf](https://github.com/sinara-hw/zapper/files/4946164/FilterZeroOffset.pdf)

Is CM choke is necessary? This one is rated only 80 V and it's hard to find 200V rated. [This](https://www.digikey.com/product-detail/en/murata-electronics/PLT10HH101150PNL/490-12821-1-ND/4422452) has 300VDC but it's very expensive 10$/channel unless we use chokes for AC power lines e.g [Schaffner RN](https://pl.mouser.com/datasheet/2/355/RN_Series-978565.pdf) which  are cheaper but bigger.

gkasprow, 2020-07-20:
CMCS make sense when we have fully symmetrical load. Piezos are referenced to GND so CMC will do more harm than help.