sinara-hw/zapper #8: schematic review

gkasprow, 2020-08-12:
- [ ] make sure the schemtics designtors do not overlap
[image]
- [ ] the module must fit within 4HP slot. the switches are too high. use smaller ones, DIP switches would do the job.
- [ ] the C51 is too high. why film capacitor was used?
- [ ] R4 won't work for 200V due to 0603 package. Use 1210 package instead or two 0805 resistors in series
- [ ] what is R9 for?
- [ ] use DIP switch as SW1
- [ ] use only CERN components. If they do not exist, let me know
- [ ] do we need AD8544 for LEDs? use LM324 or something low cost
- [ ] add 0R resistors for debugging: reference output,
- [ ] when changing he HV, adjust the amplifier gain accordingly. Add gain setting DIP switch to the DAC
- [ ] P200V0 is misleading since this value depends on SW0 setting
- [ ] this is low noise design. Are you sure the SMPS noise is sufficiently filtered by LC filter and HV amplifier PSRR for SMPS operating frequency? What is HV amplifier PSRR at such high frequency? One may need to add HV LDO.

jordens, 2020-08-14:
You can get rid of the A side switches at the input to the noise eaters.

gkasprow, 2020-09-29:
- [ ] the EEM connector is placed in random way. Please ensure CPCIS adapter compatibility