sinara-hw/zapper #9: v1.0 testing

gkasprow, 2020-09-06:
@dtcallcock @hartytp @dhslichter 
The first prototype is assembled.
What tests do you want us to perform?

dtcallcock, 2020-09-22:
How about testing the following, both with no load and a 1uF load to simulate a piezo:

- Small signal bandwidth
- Full power bandwidth
- Slew Rate
- RMS output noise with with filter
- RMS output noise without filter

Now, with 3 voltage output settings and 3 opamp bias current settings, that would be 90 distinct measurements so it might be worth picking a bias current and voltage for exhaustive testing and just do a few spot checks on other combos.

dtcallcock, 2020-09-22:
The spec sheets that piezo drive make are quite nice and could be a bit of a template: https://www.piezodrive.com/drivers/pd32-32-channel-piezo-driver/

gkasprow, 2020-09-22:
when testing, do not use ceramic capacitors because they have very strong C=f(V)